arduino-libraries/Arduino_BQ24195 #5: PMIC_IRQ_PIN' was not declared in this scope

aylashiv, 2020-09-17:
When i compile the Example of  Battery Charger in arduino, i'm getting this Error: **Arduino_BQ24195\src\BQ24195.cpp:69:15: error: 'PMIC_IRQ_PIN' was not declared in this scope**

Adk90, 2021-01-26:
I'm using MKR 1500 NB to test. Well I'm not able to charge my 1400mAh cell with more than 100mA even I do set 500mA.
What can I do to get higher charge power?

Having a look into the Lib and doing some tests, I have got the following points:

0.15 instead of 0.015  as Datasheet sais 150mA for 0x01
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L325

0.9 instead of CURRENT_LIM_900
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L331

Same here: 0.015 ?
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L368

Having a look on disableCharge() is doing more than just disable the Battery, is reseting the BMIC to default
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L209 

Thanks, Adrian

per1234, 2021-01-26:
This should be fixed by updating to the latest version of Arduino SAMD Boards via Boards Manager. Make sure to also update to the latest version of the library via Library Manager while you're at it.

closed 2021-01-26 by per1234

JMyrng, 2021-03-03:
I´m using the online Arduino editor so boards and library should be up to date. But I can´t compile the mentioned example. The error is the same: "'PMIC_IRQ_PIN' was not declared in this scope".