arduino-libraries/Arduino_BQ24195 #6: Compilation error : invalid operands of types 'double' and 'int' to binary 'operator&'

feraudet, 2020-12-18:
Hello,
I've those errors. Any idea on what's going on ?
Thanks in advance,
Cyril
```shell
Arduino: 1.8.13 (Mac OS X), Board: "Arduino MKR WiFi 1010"
...
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setInputVoltageLimit(float)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:283:88: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(INPUT_SOURCE_REGISTER, (mask | (round((voltage - 3.88) * 100) & 0x78)));
                                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setMinimumSystemVoltage(float)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:561:57: error: invalid operands of types 'byte {aka unsigned char}' and 'double' to binary 'operator|'
     return writeRegister(POWERON_CONFIG_REGISTER, (mask | ((round((voltage - 3.0) * 10) * 2) + 1)));
                                                    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setChargeCurrent(float)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:600:95: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, (round(((current - 0.512) / 0.016)) & 0xFC) | mask);
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setPreChargeCurrent(float)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:639:105: error: invalid operands of types 'float' and 'int' to binary 'operator&'
     return writeRegister(PRECHARGE_CURRENT_CONTROL_REGISTER, mask | (round((current - 0.128f) / 0.008f) & 0xF0));
                                                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setTermChargeCurrent(float)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:681:103: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(PRECHARGE_CURRENT_CONTROL_REGISTER, mask | (round((current - 0.128) / 0.128) & 0x0F));
                                                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setChargeVoltage(float)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:721:95: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(CHARGE_VOLTAGE_CONTROL_REGISTER, (round(((voltage - 3.504) / 0.004)) & 0xFC) | mask);
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setThermalRegulationTemperature(int)':
/Users/cferaudet/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:791:91: error: invalid operands of types '__gnu_cxx::__enable_if<true, double>::__type {aka double}' and 'int' to binary 'operator&'
     return writeRegister(THERMAL_REG_CONTROL_REGISTER, (mask | (round((degree - 60) / 20) & 0x03)));
                                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
exit status 1
Error compiling for board Arduino MKR WiFi 1010.
```

rengul, 2020-12-19:
Hi, 
I have the same mistake. Arduino opla MKR 1010.

Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setInputVoltageLimit(float)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:283:88: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(INPUT_SOURCE_REGISTER, (mask | (round((voltage - 3.88) * 100) & 0x78)));
                                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setMinimumSystemVoltage(float)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:561:57: error: invalid operands of types 'byte {aka unsigned char}' and 'double' to binary 'operator|'
     return writeRegister(POWERON_CONFIG_REGISTER, (mask | ((round((voltage - 3.0) * 10) * 2) + 1)));
                                                    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setChargeCurrent(float)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:600:95: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, (round(((current - 0.512) / 0.016)) & 0xFC) | mask);
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setPreChargeCurrent(float)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:639:105: error: invalid operands of types 'float' and 'int' to binary 'operator&'
     return writeRegister(PRECHARGE_CURRENT_CONTROL_REGISTER, mask | (round((current - 0.128f) / 0.008f) & 0xF0));
                                                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setTermChargeCurrent(float)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:681:103: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(PRECHARGE_CURRENT_CONTROL_REGISTER, mask | (round((current - 0.128) / 0.128) & 0x0F));
                                                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setChargeVoltage(float)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:721:95: error: invalid operands of types 'double' and 'int' to binary 'operator&'
     return writeRegister(CHARGE_VOLTAGE_CONTROL_REGISTER, (round(((voltage - 3.504) / 0.004)) & 0xFC) | mask);
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp: In member function 'bool PMICClass::setThermalRegulationTemperature(int)':
/Users/gulinellire/Documents/Arduino/libraries/Arduino_BQ24195/src/BQ24195.cpp:791:91: error: invalid operands of types '__gnu_cxx::__enable_if<true, double>::__type {aka double}' and 'int' to binary 'operator&'
     return writeRegister(THERMAL_REG_CONTROL_REGISTER, (mask | (round((degree - 60) / 20) & 0x03)));

facchinm, 2020-12-21:
Hi @feraudet and @rengul ,
the issue is due to new samd core (1.8.10).
Older code used this `round()` definition 
https://github.com/arduino/ArduinoCore-samd/blob/b9b84b6850aee887c36bb8de9c0942f3eff2b617/cores/arduino/Arduino.h#L104
while the new one uses standard C++ (http://www.cplusplus.com/reference/cmath/round/) which returns a `double`.
The workaround while we fix the issue is either downgrading to core 1.8.9 or adding this line at the beginning of `BQ24195.cpp`
```
#define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
```
@cmaglie @aentinger any proposal on how to fix this properly?

rengul, 2020-12-21:
Hi @facchinm 
adding line at the beginning now work. 
Thanks for the reply.

agf-lab, 2020-12-30:
we experiece and apparently fix the same issue by casting (byte) on each "round" call . i.e. patching :
     return writeRegister(INPUT_SOURCE_REGISTER, (mask | (round((voltage - 3.88) * 100) & 0x78)));
with :
    return writeRegister(INPUT_SOURCE_REGISTER, (mask | (**(byte)**(round((voltage - 3.88) * 100)) & 0x78)));
hoping the round((voltage - 3.88) * 100) < 255

closed 2020-12-30 by facchinm