arduino-libraries/Arduino_BQ24195 #7: Current Setting and Disable Charging, unable to charge higher than 100mA

Adk90, 2021-01-26:
I'm using MKR 1500 NB to test. Well I'm not able to charge my 1400mAh cell with more than 100mA even I do set 500mA.
What can I do to get higher charge power?

Having a look into the Lib and doing some tests, I have got the following points:

0.15 instead of 0.015  as Datasheet sais 150mA for 0x01
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L325

0.9 instead of CURRENT_LIM_900
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L331

Same here: 0.015 ?
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L368

Having a look on disableCharge() is doing more than just disable the Battery, is reseting the BMIC to default, guess this should be mentioned in the description if by intention.
https://github.com/arduino-libraries/Arduino_BQ24195/blob/46bc843614d059e9f58bba0248b0453a765db212/src/BQ24195.cpp#L209 

Thanks, Adrian

MCXIV, 2021-02-18:
Hi Adrian, have you find a solution to charge more than 100 mA by any chance? 

I found a way to reach ~150 mA (without the library) but I'd like to charge at 500 mA.

Adk90, 2021-02-18:
no, I did not investigate future.

MCXIV, 2021-02-19:
Hi Adrian, for your info, here's a bit of code to charge at 450/500 mA. It doesn't use this library tho.

```c++
#include <Wire.h>

#define BBQ_addr 0x6B

void setup() {
  Serial.begin(9600);
  Wire.begin();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  Serial.println(isUSBconnected());
}

int isUSBconnected() {
  Wire.beginTransmission(BBQ_addr);
  Wire.write((byte)0x08);
  Wire.endTransmission();

  Wire.beginTransmission(BBQ_addr);
  Wire.requestFrom(BBQ_addr, 1);
  Wire.endTransmission();
  
  char binary[9] = {0};
  itoa(Wire.read() + 256, binary, 2);
  if (binary[3] == '1' && binary[4] == '0') {
    Wire.beginTransmission(BBQ_addr);
    Wire.write((byte)0x00); // Register 0
    Wire.write((byte)0x36); // Set I_lim @ 500 mA
    Wire.endTransmission();
    digitalWrite(LED_BUILTIN, HIGH);
    return 1;
  }
  else {
    digitalWrite(LED_BUILTIN, LOW);
    return 0;
  }
}
```
It sets the limitation at 500 mA when the USB is connected. Don't put this code in the *setup*, it will not work due to a strange PCB designing : Arduino left the D+/D- pin of the BQ24195 floating, causing the CI to think that there is not USB connected. Very strange and somewhat dangerous in some cases I guess.

Cheers!
Quentin

Adk90, 2021-02-22:
Thanks, once I'll grap the Borad again, I'll check.

LucasISC, 2021-03-10:
Hi, I am also trying to charge a battery connected to a MKR1010 board and encountered the same issue when trying the BQ24195 library (only charging at 100mA). 
I tested your code @MCXIV and managed to get it work with a current >300 mA which is a progress. Unfortunately I am now trying to make it work with higher current (around 2.0A or so) but despite testing a lot of setting and playing with the registers it doesn't change anything. Do you have an idea how to manage this ?

Thanks!
Lucas

MCXIV, 2021-03-10:
Hey @LucasISC, can you tell me a little bit more about your setup? What is connected to your MKR, which battery are you using, are you charging from a usb port of your computer, ...?

Unfortunately, you cannot go up to 2A. Yes the datasheet says so, but the fuse limits the current at (In theory) 500 mA. I was able to charge at almost 800 mA but the fuse was extremely hot.

I do not recommend it but, you can try :

```py
Wire.beginTransmission(BBQ_addr);
Wire.write((byte)0x02);
Wire.write((byte)0x80); 
Wire.endTransmission();
```
Just switched from 0x00 to 0x80, it should set the Charge Current Control at 2048 mA instead of 512 mA

Hope it helps, do not hesitate to ask me again, I'm learning everyday from this IC haha.

Cheers!
Quentin

LucasISC, 2021-03-11:
Hi, thanks for the quick reply! I am using a MKR1010 and I am trying to charge a 1S3P Li-ion battery (3.7V, 7800mAh). I connected the arduino to the computer and to a charger but the results are the same.
Ok now I understand why it doesn't go higher than 500mA, I even tested by setting the Charge Current Control at the maximum (around 4.0A I think) without results.
I will stick with my previous configuration using a tp4056 module to charge that battery haha

Thanks again for the help!

Regards
Lucas

MCXIV, 2021-04-26:
For anyone lurking this issue in the hope of controlling the input charge current easily, here is a bit of code for you.
It works fine on a MKR1010, I strongly suggest not to charge at more than 500 mA as the security fuse on the MKR1010 is given for a hold current of 500 mA. 

I also suggest not to charge at 500 mA as it eff up the ADC for the battery voltage monitor on the D21G18A. (At least, on my board)

**This code only works for the fast charging part!**

**The function *BQ()* cannot be use in the *setup()***

```cpp
#include <Wire.h>

int start = 0;

void setup() {
  Wire.begin();
}

void loop() {
  if (start == 0) {
    BQ(0x6b, 0x32, 0x61);
    start = 1;
  }
}

void BQ(byte addr, byte a, byte b) {
  Wire.beginTransmission(addr);
  Wire.write(0x00);
  Wire.write(a); // Input current limit (In our example, 0x32 is 500 mA)
  Wire.endTransmission();
  /////
  Wire.beginTransmission(addr);
  Wire.write(0x02);
  Wire.write(b); // Charge current (In our example, 0x61 is 300 mA)
  Wire.endTransmission();
}
```
[image]

Just a bit of explanation with the datasheet ; what we're doing is that we change the register n°2 on the BQ24195 which is the register to change our charge current. 

We set the bit n°0 at 1 to have a better sampling in our current charge. (If we let it at 0, by default, you cannot set a charge current at less than 512 mA)

Bit n°2 to Bit n°7 will be our future charge current. Just add whatever value and then divide the result by 5 : here is our charge current !

### Example :

If we want to select 300 mA for our charge current, we can set the bit n°6 and 5 at 1 **->** (1024 mA + 512 mA) / 5 = ~307 mA)
So, our register n°2 will be **0110 0001** which is 0x61 in HEXA.

Hope it helps someone one day. :)

ShadF0x, 2022-04-06:
It says in the doc that the default charge current is 2048 mA, and if you set all bits to 0, you still have 512 mA.

So the default 0x60 is: 
1024 mA [bit 6] + 512 mA [bit 5] + **512 mA** [base] = 2048 mA

If you want ~300 mA, you should use 0x41 (disable 5th bit).

oscarfallman, 2022-09-13:
Hi @MCXIV, I'm a bit late to the party but I'm struggling with this ATM. Why can't the code you posted be run during setup()? I'm having an issue where i need to set the Input current limit during setup, and i wanted to use your method. Thanks!

MCXIV, 2022-09-27:
> Hi @MCXIV, I'm a bit late to the party but I'm struggling with this ATM. Why can't the code you posted be run during setup()? I'm having an issue where i need to set the Input current limit during setup, and i wanted to use your method. Thanks!

Hello @oscarfallman!

Sorry, it's been so long, I can't remember. Actually, I think I never knew why. The best solution for you would be https://github.com/arduino-libraries/Arduino_BQ24195/issues/7#issuecomment-826715065 I guess.

Let me know if you find something else. ;) 
Quentin

alexsee75, 2023-03-26:
Just wanted to point out: The easiest way to enable charging current of > 100mA is to deactivate DPDM (D+D-) detection using disableDPDM() from this library. E.g. something like this will work:

``
 if (PMIC.begin()) {
    PMIC.disableDPDM();
    PMIC.end();
  }
``

You can also use setChargeCurrent() but it's already set to 2A while thermal constraints limit the charging current to around 0,5A. Adding a heatsink to the BQ24195 module will probably improve this to around +/- 1A (although not 2A)

Disadvantage is that every plug/unplug of USB (perhaps battery as well, did not check) will reactivate DPDM so you will have to rerun this then. In my case I run it on restart and in each main loop (in between deep sleep) which seems to work and will enable reasonable charging (with some delay due to the next time the loop is run). I've not found a way to permanently fix it and according to datasheet it's IMHO not possible....

The best way would be for Arduino to connect D+/D- properly, or at least run a 200 ohm resistor between them which would always force a charging port detection. It's probably not feasible to modify the board since the connections are so small.

Best,
  Alex

geludwig, 2023-07-18:
Hello to you all.

I have bought an Arduino MKR NB 1500 this June and have ran into some problems with it.

Nevertheless, here are some results for the BQ24195 library and the MKR NB 1500. I modified the _setChargeCurrent()_ function for this, because it cant set currents below 512mA (current gets overwritten with 512mA when below). Please correct me if I am wrong!!!

I disabled the V+ Pin of the USB connection and powered the board directly via the VIN pin. I tried powering the board via USB only, but that resulted in bootloops of the SAMD21 when initializing the SARA-R4 module. The whole board seems to bit unstable when using no battery and/or no perfect input voltage.

`/*******************************************************************************
 * Function Name  : setChargeCurrent
 * Description    : Set charge current
 * Input          : The current expressed in Ampere
 * Return         : 0 on Error, 1 on Success
 *******************************************************************************/
bool PMICClass::setChargeCurrent(float current) {
    int DATA = readRegister(CHARGE_CURRENT_CONTROL_REGISTER);

    if (DATA == -1) {
        return 0;
    }
    byte mask = DATA & 0x01;

    if (current > 4.544) {
        current = 4.544;
    } else if (current < 0.512) {
        current = 0.512;
    }
	
	/*
	0x01 = 100mA (measured 100mA)
	0x05 = 115mA (measured 115mA)
	0x09 = 130mA (measured128mA)
	0x11 = 150mA (measured 155mA)
	0x21 = 200mA (measured 180mA)
	0x41 = 300mA (measured 285mA)
	*/
	
	writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, 0x41);
	return 1;
	
	//return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, (round(((current - 0.512) / 0.016)) & 0xFC) | mask);`

The following code was used for initialization.
`  while (PMIC.begin() == 0)
  {
    u8x8.drawString(0,4,"INIT PMIC FAIL");
  }
  if ((PMIC.setFastChargeTimerSetting(pmicHours) == 1) &&
      (PMIC.setChargeVoltage(batFullVoltage) == 1) &&
      (PMIC.setMinimumSystemVoltage(batEmptyVoltage) == 1) && 
      (PMIC.setChargeCurrent(batChargeCurrent) == 1) &&
      (PMIC.enableCharge() == 1))
  {
    u8x8.clearLine(4);
    u8x8.drawString(0,4,"PMIC OK");`

alexsee75, 2023-07-21:
Hello geludwig,

Concerning the SARA-R4 Module: you should update it to the latest
version L.05.11 A.02.16. Arduino support has a two-step process
which just needs you to update the SerialSARA sketch to enable a
direct serial connection to the SARA-R4 module, plus a serial
terminal with X-modem file transfer capability (I used minicom under
Linux). No soldering necessary. Without the update, the module is
in fact very unstable - after the update it worked reasonably well.

IMHO - provided the MKR 1500 NB has the same USB connections as the
MKR 1010 Wifi - a connection with USB will force 100mA maximum
charge current since D+ and D- are left floating which is so
interpreted by BQ24195... this can only be "fixed" by rebooting
which gets a higher charge current.

Although I also remember there is a slight bug in the original function
for setting the current (setChargeCurrent) in the BQ24195 library. I
thihnk for low currents there is one "if" that checks a wrong
threshold. I noted it some time ago but did not have time to send in
a patch yet. Please check this, it should be easy to fix. If that's
it I'll be happy to send in a patch sometime next week.

Best,
  Alex

geludwig, 2023-07-22:
Hi Alex,
I updated the SARA-R4 module already. I soldered the USB cable directly to the board and used EasyFlash. Still a bit unstable without a battery. I made a custom daughterboard PCB with a dedicated power supply and its fine now.

Here is an example which **_could_** work.
I dont know why it would be necessary to read the register first, do a bitwise AND operation and a bitwise OR operation to the return value afterwards. We do know what bits have to be set to set a specific current, we dont need the current register status. Or am I wrong?

```
bool PMICClass::setChargeCurrent(float current) {
    int DATA = readRegister(CHARGE_CURRENT_CONTROL_REGISTER);

    if (DATA == -1) {
        return 0;
    }
	
    byte mask = DATA & 0x01;

    if (current > 4.544) {
        current = 4.544;
        return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, (round(((current - 0.512) / 0.016)) & 0xFC) | mask);
    } else if (current < 0.103) {
        current = 0.103;
        return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, round(((current * 5) - 0.512) / 0.016) | 0x01);
    }

}
```

I did a quick test and also measured the current, and it seems to work.

geludwig, 2023-07-22:
Hi Alex,

I already updated the SARA-R4 module with a soldered USB cable and EasyFlash. I designed and ordered a custom PCB daughterboard with a dedicated power supply, now its fine.

Here is the updated _setChargeCurrent()_function. I did some tests and measured the corresponding current, seems fine to me.

```
bool PMICClass::setChargeCurrent(float current) {
    int DATA = readRegister(CHARGE_CURRENT_CONTROL_REGISTER);

    if (DATA == -1) {
        return 0;
    }
	
    byte mask = DATA & 0x01;

    if (current > 4.544) {
        current = 4.544;
    } else if (current < 0.103) {
        current = 0.103;
    }

	if (current >= 0.512) {
		return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, (round((current - 0.512) / 0.016) & 0xFC) | mask);
	}
	else if (current < 0.512) {
		return writeRegister(CHARGE_CURRENT_CONTROL_REGISTER, (round(((current * 5) - 0.512) / 0.016) & 0xFD) | 0x01);
	}
}
```

cooper-david-a, 2024-12-25:
@alexsee75 could you point me toward some information about how you updated the firmware without the additional USB cable?